robertklep/esphome-delta-solivia #1: Error: 'make_unique' is not a member of 'std'

hkrogh, 2025-02-27:
Hi!

I have created the yaml and it is validating correctly. However when I compile/install I get an compile error.

ESPHOME 2025.2.1 on Home assistant 2025.2.5.

Based on the error it seems it is compiling using standard C++ but where the code requires c++ 14.0.
I have tried adding following to yaml without success:

esphome:
  platformio_options:
    build_flags: 
      - -std=c++14

Any ideas?

Btw the link to the example yaml in the read.me is invalid: https://github.com/robertklep/esphome-delta-solivia/blob/v2.0.1/esphome-example-configuration.yaml

INFO ESPHome 2025.2.1
INFO Reading configuration /config/esphome/esp32.yaml...
INFO Generating C++ source...
WARNING — [Solivia] Fixing component update interval to 500ms
INFO Compiling app...
Processing esphome-web-cb5b5c (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
Compiling .pioenvs/esphome-web-cb5b5c/src/esphome/components/delta_solivia/delta-solivia-component.cpp.o
Compiling .pioenvs/esphome-web-cb5b5c/src/esphome/components/delta_solivia/delta-solivia-crc.cpp.o
Compiling .pioenvs/esphome-web-cb5b5c/src/esphome/components/delta_solivia/delta-solivia-inverter.cpp.o
Compiling .pioenvs/esphome-web-cb5b5c/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/esphome-web-cb5b5c/src/esphome/components/esphome/ota/ota_esphome.cpp.o
In file included from src/esphome/components/delta_solivia/delta-solivia-inverter.h:12,
                 from src/esphome.h:24,
                 from src/esphome/components/delta_solivia/delta-solivia-component.h:5,
                 from src/esphome/components/delta_solivia/delta-solivia-component.cpp:1:
src/esphome/components/delta_solivia/frame-parser.h: In static member function 'static esphome::delta_solivia::FrameParser esphome::delta_solivia::FrameParserFactory::createParser(uint8_t)':
src/esphome/components/delta_solivia/frame-parser.h:19:21: error: 'make_unique' is not a member of 'std'
         return std::make_unique<FrameParserVariant15>();
                     ^~~~~~~~~~~
src/esphome/components/delta_solivia/frame-parser.h:19:21: note: 'std::make_unique' is only available from C++14 onwards
src/esphome/components/delta_solivia/frame-parser.h:19:53: error: expected primary-expression before '>' token
         return std::make_unique<FrameParserVariant15>();
                                                     ^
src/esphome/components/delta_solivia/frame-parser.h:19:55: error: expected primary-expression before ')' token
         return std::make_unique<FrameParserVariant15>();
                                                       ^
src/esphome/components/delta_solivia/frame-parser.h:22:21: error: 'make_unique' is not a member of 'std'
         return std::make_unique<FrameParserVariant212>();
                     ^~~~~~~~~~~
src/esphome/components/delta_solivia/frame-parser.h:22:21: note: 'std::make_unique' is only available from C++14 onwards
src/esphome/components/delta_solivia/frame-parser.h:22:54: error: expected primary-expression before '>' token
         return std::make_unique<FrameParserVariant212>();
                                                      ^
src/esphome/components/delta_solivia/frame-parser.h:22:56: error: expected primary-expression before ')' token
         return std::make_unique<FrameParserVariant212>();
                                                        ^
In file included from src/esphome/components/delta_solivia/delta-solivia-inverter.h:12,
                 from src/esphome.h:24,
                 from src/esphome/components/delta_solivia/delta-solivia-crc.h:1,
                 from src/esphome/components/delta_solivia/delta-solivia-crc.cpp:2:
src/esphome/components/delta_solivia/frame-parser.h: In static member function 'static esphome::delta_solivia::FrameParser esphome::delta_solivia::FrameParserFactory::createParser(uint8_t)':
src/esphome/components/delta_solivia/frame-parser.h:19:21: error: 'make_unique' is not a member of 'std'
         return std::make_unique<FrameParserVariant15>();
                     ^~~~~~~~~~~
src/esphome/components/delta_solivia/frame-parser.h:19:21: note: 'std::make_unique' is only available from C++14 onwards
src/esphome/components/delta_solivia/frame-parser.h:19:53: error: expected primary-expression before '>' token
         return std::make_unique<FrameParserVariant15>();
                                                     ^
src/esphome/components/delta_solivia/frame-parser.h:19:55: error: expected primary-expression before ')' token
         return std::make_unique<FrameParserVariant15>();
                                                       ^
src/esphome/components/delta_solivia/frame-parser.h:22:21: error: 'make_unique' is not a member of 'std'
         return std::make_unique<FrameParserVariant212>();
                     ^~~~~~~~~~~
src/esphome/components/delta_solivia/frame-parser.h:22:21: note: 'std::make_unique' is only available from C++14 onwards
src/esphome/components/delta_solivia/frame-parser.h:22:54: error: expected primary-expression before '>' token
         return std::make_unique<FrameParserVariant212>();
                                                      ^
src/esphome/components/delta_solivia/frame-parser.h:22:56: error: expected primary-expression before ')' token
         return std::make_unique<FrameParserVariant212>();
                                                        ^
In file included from src/esphome.h:29,
                 from src/esphome/components/delta_solivia/delta-solivia-inverter.h:7,
                 from src/esphome/components/delta_solivia/delta-solivia-inverter.cpp:1:
src/esphome/components/delta_solivia/frame-parser.h: In static member function 'static esphome::delta_solivia::FrameParser esphome::delta_solivia::FrameParserFactory::createParser(uint8_t)':
src/esphome/components/delta_solivia/frame-parser.h:19:21: error: 'make_unique' is not a member of 'std'
         return std::make_unique<FrameParserVariant15>();
                     ^~~~~~~~~~~
src/esphome/components/delta_solivia/frame-parser.h:19:21: note: 'std::make_unique' is only available from C++14 onwards
src/esphome/components/delta_solivia/frame-parser.h:19:53: error: expected primary-expression before '>' token
         return std::make_unique<FrameParserVariant15>();
                                                     ^
src/esphome/components/delta_solivia/frame-parser.h:19:55: error: expected primary-expression before ')' token
         return std::make_unique<FrameParserVariant15>();
                                                       ^
src/esphome/components/delta_solivia/frame-parser.h:22:21: error: 'make_unique' is not a member of 'std'
         return std::make_unique<FrameParserVariant212>();
                     ^~~~~~~~~~~
src/esphome/components/delta_solivia/frame-parser.h:22:21: note: 'std::make_unique' is only available from C++14 onwards
src/esphome/components/delta_solivia/frame-parser.h:22:54: error: expected primary-expression before '>' token
         return std::make_unique<FrameParserVariant212>();
                                                      ^
src/esphome/components/delta_solivia/frame-parser.h:22:56: error: expected primary-expression before ')' token
         return std::make_unique<FrameParserVariant212>();
                                                        ^
*** [.pioenvs/esphome-web-cb5b5c/src/esphome/components/delta_solivia/delta-solivia-crc.cpp.o] Error 1
*** [.pioenvs/esphome-web-cb5b5c/src/esphome/components/delta_solivia/delta-solivia-inverter.cpp.o] Error 1
*** [.pioenvs/esphome-web-cb5b5c/src/esphome/components/delta_solivia/delta-solivia-component.cpp.o] Error 1
========================= [FAILED] Took 14.51 seconds =========================

hkrogh, 2025-02-27:
I added following buildflags instead and now it compiles.

esphome:
  platformio_options:
    build_flags: "-std=gnu++17"
    build_unflags: "-std=gnu++11"

It is working fine with two RPI H3 (variant 212) inverters - which I had running with a custom component already. I just named the entities as my old setup, so all entities kept their names.

Great work!

closed 2025-02-27 by hkrogh

robertklep, 2025-02-27:
Thanks! I actually use the same build flags in my own code, but I forgot to add them to the example YAML file 🥴

commit e8b766fb (2025-02-27): Set correct build flags (fixes #1)