robertklep/esphome-delta-solivia #3: [18:47:18][D][DeltaSolivia:028]: FRAME - incorrect header

fontexD, 2025-05-05:
Im using a G3 eu so variant 18

i keep getting this 

[18:47:18][D][DeltaSolivia:028]: FRAME - incorrect header
[18:47:28][D][DeltaSolivia:052]: INVERTER1 - requesting update
[18:47:28][D][DeltaSolivia:028]: FRAME - incorrect header
[18:47:38][D][DeltaSolivia:052]: INVERTER1 - requesting update
[18:47:38][D][DeltaSolivia:028]: FRAME - incorrect header

as of it cant connect to it, what could i be doing wrong ? 
its fine connected to wifi etc, turned on debug mode aswell, 

Used this setup

`esphome:
  name: 'delta-solivia-reader'
  friendly_name: 'Delta Solivia Reader'
  platformio_options:
    build_unflags: '-std=c++11 -std=gnu++11'
    build_flags: '-std=c++17 -std=gnu++17'

external_components:
  - source: github://robertklep/esphome-delta-solivia@v3.1.0
    components: [ delta_solivia ]

esp32:
  # look here for more information on board ID's:
  # https://esphome.io/components/esp32.html#esp32-platform
  board: esp32dev

# enable logging (entirely optional)
logger:
  level: DEBUG
  baud_rate: 115200

# enable wifi
wifi:
  ssid: "<redacted>"
  password: "<redacted>"

# enable Home Assistant API
api:

# enable over-the-air updates
ota:
  - platform: esphome

# configure a UART (required!)
uart:
  id: solivia_uart
  tx_pin: GPIO2
  rx_pin: GPIO4
  rx_buffer_size: 512
  baud_rate: 19200
  parity: NONE
  stop_bits: 1

# configure the Delta Solivia component
delta_solivia:
  uart_id: solivia_uart       # the id of the UART configured above
  flow_control_pin: GPIO5     # see README.md
  has_gateway: false          # see README.md
  update_interval: 10s        # see README.md
  inverters:
    - address: 1
      variant: 18
      throttle: 30s           # see README.md
      part_number:
        name: 'Inverter#1 Part Number'
      serial_number:
        name: 'Inverter#1 Serial Number'
      ac_power:
        name: 'Inverter#1 Current Power'
      supplied_ac_energy_total:
        name: 'Inverter#1 Total Energy'
      supplied_ac_energy_today:
        name: 'Inverter#1 Today Energy'
        filters:
          - timeout: 2h
      solar_voltage_input_1:
        name: 'Inverter#1 Solar Voltage Input 1'
      solar_current_input_1:
        name: 'Inverter#1 Solar Current Input 1'
      ac_voltage:
        name: 'Inverter#1 AC Voltage'
      ac_current:
        name: 'Inverter#1 AC Current'
      ac_frequency:
        name: 'Inverter#1 AC Frequency'
      ac_grid_voltage:
        name: 'Inverter#1 AC Grid Voltage'
      ac_grid_frequency:
        name: 'Inverter#1 AC Grid Frequency'
      runtime_total:
        name: 'Inverter#1 Runtime Total'
      runtime_today:
        name: 'Inverter#1 Runtime Today'
      max_ac_power_today:
        name: 'Inverter#1 Max AC Power Today'
      max_solar_power_input_1:
        name: 'Inverter#1 Max Solar Power Input 1'

robertklep, 2025-05-05:
Which model do you have specifically?

robertklep, 2025-05-05:
Are you sure the hardware is set up correctly?

robertklep, 2025-05-06:
It looks like the flow control is connected to both GPIO0 and GPIO5? I'm not sure if that might cause issues, GPIO0 often has a special purpose on ESP32s.

If disconnecting from GPIO0 doesn't fix anything, try swapping TX and RX (either the wires, or in the YAML file, but not both).

fontexD, 2025-05-06:
Ye we removed it so its on 5 now

is the wrigin correct with Green on pin 6 andpin 7 white/brown ?

robertklep, 2025-05-06:
Yes, the colors sound okay.

The log suggests that the component is trying to send a request to the bus, but it doesn't receive any response.

You can try this:
```
build_flags: '-std=c++17 -std=gnu++17 -DDEBUG_HEADER'
```

It should log additional header data that might be useful in tracking down this issue.

fontexD, 2025-05-06:
ddint change anything it seems
tried with and withtout gateway on, 
we are using a Wroom 32D
and a rs485 ttl https://ardustore.dk/produkt/max485-til-rs485-converter-adapter-module
checked cables a hundred times, also veried with you they were correct earlier
tried with and withtout flowcontrol connected
D1 lights up but no other led

https://ibb.co/n8fQhrYJ
https://ibb.co/pjqpzYmJ
really confused about this tbh, should just work outta da box 

esphome:
  name: 'delta-solivia-reader'
  friendly_name: 'Delta Solivia Reader'
  platformio_options:
    build_unflags: '-std=c++11 -std=gnu++11'
    build_flags: '-std=c++17 -std=gnu++17 -DDEBUG_HEADER'

external_components:
  - source: github://robertklep/esphome-delta-solivia@v3.1.0
    components: [ delta_solivia ]

esp32:
  # look here for more information on board ID's:
  # https://esphome.io/components/esp32.html#esp32-platform
  board: esp32dev

# enable logging (entirely optional)
logger:
  level: VERBOSE
  baud_rate: 0

# enable wifi
wifi:
  ssid: "Fontex_iot"
  password: "<redacted>"

# enable Home Assistant API
api:

# enable over-the-air updates
ota:
  - platform: esphome

# configure a UART (required!)
uart:
  id: solivia_uart
  tx_pin: GPIO1
  rx_pin: GPIO3
  rx_buffer_size: 512
  baud_rate: 19200
  parity: NONE
  stop_bits: 1

# configure the Delta Solivia component
delta_solivia:
  uart_id: solivia_uart       # the id of the UART configured above
  flow_control_pin: GPIO22    
  has_gateway: true          # see README.md
  update_interval: 10s        # see README.md
  inverters:
    - address: 1
      variant: 18
      throttle: 30s           # see README.md
      part_number:
        name: 'Inverter#1 Part Number'
      serial_number:
        name: 'Inverter#1 Serial Number'
      ac_power:
        name: 'Inverter#1 Current Power'
      supplied_ac_energy_total:
        name: 'Inverter#1 Total Energy'
      supplied_ac_energy_today:
        name: 'Inverter#1 Today Energy'
        filters:
          - timeout: 2h
      solar_voltage_input_1:
        name: 'Inverter#1 Solar Voltage Input 1'
      solar_current_input_1:
        name: 'Inverter#1 Solar Current Input 1'
      ac_voltage:
        name: 'Inverter#1 AC Voltage'
      ac_current:
        name: 'Inverter#1 AC Current'
      ac_frequency:
        name: 'Inverter#1 AC Frequency'
      ac_grid_voltage:
        name: 'Inverter#1 AC Grid Voltage'
      ac_grid_frequency:
        name: 'Inverter#1 AC Grid Frequency'
      runtime_total:
        name: 'Inverter#1 Runtime Total'
      runtime_today:
        name: 'Inverter#1 Runtime Today'
      max_ac_power_today:
        name: 'Inverter#1 Max AC Power Today'
      max_solar_power_input_1:
        name: 'Inverter#1 Max Solar Power Input 1'

robertklep, 2025-05-07:
When `-DDEBUG_HEADER` the logs should look like this:
```
[06:56:13][D][DeltaSolivia:052]: INVERTER2 - requesting update
[06:56:13][D][DeltaSolivia:069]: FRAME - SOP = 0x02, PC = 0x06, address = 1, data size = 150, cmd = 0x60, sub cmd = 0x01
```

In your case, it's only showing "requesting update" but then nothing, which suggests it's simply not receiving anything.

I just posted [this](https://community.home-assistant.io/t/rs-485-component-for-some-delta-solivia-g3-g4-solar-inverters/788218/33) on the HA forum, with someone else having a similar issue. It could be that you need to turn off RS-485 termination before any data gets passed.

I never had to do this myself because my system had a gateway connected to it when I got it, hence it's also not documented (if this turns out to be the cause, I'll update the README).

fontexD, 2025-05-07:
When using a gateway how do you connect it ? Gateway in one rs485 and esp in the other rs485 on the inverter or ? We also have a gateway laying Sent from Android device7.

robertklep, 2025-05-07:
I don't use the gateway anymore, and it's also not a requirement. It's just that because my system came with a gateway, its termination may have already been turned off.

fontexD, 2025-05-09:
Is there any way we could debug the data coming in or out? Just to see if anything happens at all

robertklep, 2025-05-09:
The data going in and out is already being logged:
```
[19:23:05][D][DeltaSolivia:052]: INVERTER1 - requesting update
[19:23:05][D][uart_debug:114]: >>> 02:05:01:02:60:01:85:FC:03
[19:23:05][D][DeltaSolivia:028]: FRAME - incorrect header
[19:23:05][W][component:239]: Component delta_solivia took a long time for an operation (86 ms).
[19:23:05][W][component:240]: Components should block for at most 30 ms.
[19:23:05][D][uart_debug:114]: <<< 00
```
* `>>>` is the data going out, which is the component requesting an update
* `<<<` is the data coming in, which seems to be only a single NUL byte

Although it seems strange that it's not actually logging anything in between the component requesting an update and then logging "incorrect header"; there should be data being received in between.

We can try enabling more UART debugging:
```
uart:
  id: solivia_uart
  tx_pin: GPIO2
  rx_pin: GPIO4
  rx_buffer_size: 512
  baud_rate: 19200
  parity: NONE
  stop_bits: 1
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      bytes: 32
    sequence:
      - lambda: UARTDebug::log_hex(direction, bytes, ':');
```

fontexD, 2025-05-09:
[image]

fontexD, 2025-05-09:
i also have one of these ?  idk if i could test anything with it and connect to pc ? 

https://el-grossisten.dk/shop/h59506-usb-to-rs485-interface-converter-ttl-38502

robertklep, 2025-05-09:
"RESPONSE - timeout" means that your inverter didn't respond within the correct time. But it now looks like there is no data coming in at all, otherwise it would get logged by the UART debug code.

You can try to use the USB device from a PC to see if the inverter is sending any data on the bus by itself, but I'm not sure it does (in other words, it needs to be explicitly asked before it transmits anything).

robertklep, 2025-05-09:
My inverters came with the house when I bought it, and because they were already connected to a gateway I never had to set up the inverters myself, but I wonder if there's perhaps an inverter setting required before it actually uses the RS-485 bus at all.

robertklep, 2025-08-09:
@JM85Z thanks! It may take a few days for me to be able to take a closer look at it.

JM85Z, 2025-08-13:
Hi @robertklep , i only copied the uart_debug data. I think most of the received data ends in a connection error (and a restart). 
I don't get any readings on the sensors in HA, so it seems every uart_debug data i receive from the does not get parsed. 

I can give you a full log this evening.

JM85Z, 2025-08-16:
In attachment a complete log. One time i had a "incorrect frame size".
Maybe this can help? 

[solivia errors 2.txt](https://github.com/user-attachments/files/21817782/solivia.errors.2.txt)

robertklep, 2025-08-17:
Thanks! That's a very useful log.

For some reason, there can be more data available in the UART buffer than expected. I didn't know that was possible, but apparently it is and it's causing a buffer overrun which is causing crashes.

I'll have to rewrite the packet handling to see if I can only read the actual require amount of data for a frame.

robertklep, 2025-08-17:
@JM85Z can you try this?

```
external_components:
  - source: github://robertklep/esphome-delta-solivia@v3.3.0-beta.1
    components: [ delta_solivia ]
```

JM85Z, 2025-08-17:
@robertklep I just updated the esp32 and it's reading data from the inverter. So it's looking ok! I will check the value's during the coming days and keep you updated. Thank you very much for this debugging.

robertklep, 2025-08-30:
I just pushed v3.4.0, which contains these fixes but also another one that was causing issues with sensors not getting updated properly (#6). It should perform much better now:
```
external_components:
  - source: github://robertklep/esphome-delta-solivia@v3.4.0
    components: [ delta_solivia ]
```

closed 2025-08-30 by robertklep