robertklep/esphome-delta-solivia #4: Addition for limiting inverter output

NickyNL, 2025-06-14:
First of all let me say I`m very happy I found this project!
The earlier two projects that worked similarly, had stopped working since esphome had stopped supporting some functions,
And I was unable after a few hours of tinkering to get those others working again.

I do though have a possible worthy addition for your project;
I scanned which codes pop up when the gateway is set to limit the inverters output to 0,30 or 60% of its output.
Now I am able to send this also via this esphome device.
Eliminating the total need for the gateway, and for relays to control that gateway.

Might be usefull to add this to your project, for yourself or for others.

I have done it like this:
(Not expert with coding, so I`m also open to improvements, but it does work)

switch:
  - platform: template
    name: "UGTI Shed 0%"
    id: ugti_shed_0
    optimistic: true
    turn_on_action:
      - switch.turn_off: ugti_shed_30
      - delay: 500ms
      - switch.turn_off: ugti_shed_60
      - delay: 500ms
      - uart.write:
          id: solivia_uart
          data: [0x02, 0x05, 0xFF, 0x03, 0x0D, 0x88, 0x00, 0xE3, 0xC6, 0x03]
    turn_off_action:
      - uart.write:
          id: solivia_uart
          data: [0x02, 0x05, 0xFF, 0x03, 0x0D, 0x88, 0x64, 0xE2, 0x2D, 0x03]
  - platform: template
    name: "UGTI Shed 30%"
    id: ugti_shed_30
    optimistic: true
    turn_on_action:
      - switch.turn_off: ugti_shed_0
      - delay: 500ms
      - switch.turn_off: ugti_shed_60
      - delay: 500ms
      - uart.write:
          id: solivia_uart
          data: [0x02, 0x05, 0xFF, 0x03, 0x0D, 0x88, 0x1E, 0x63, 0xCE, 0x03]
    turn_off_action:
      - uart.write:
          id: solivia_uart
          data: [0x02, 0x05, 0xFF, 0x03, 0x0D, 0x88, 0x64, 0xE2, 0x2D, 0x03]
  - platform: template
    name: "UGTI Shed 60%"
    id: ugti_shed_60
    optimistic: true
    turn_on_action:
      - switch.turn_off: ugti_shed_0
      - delay: 500ms
      - switch.turn_off: ugti_shed_30
      - delay: 500ms
      - uart.write:
          id: solivia_uart
          data: [0x02, 0x05, 0xFF, 0x03, 0x0D, 0x88, 0x3C, 0xE3, 0xD7, 0x03]
    turn_off_action:
      - uart.write:
          id: solivia_uart
          data: [0x02, 0x05, 0xFF, 0x03, 0x0D, 0x88, 0x64, 0xE2, 0x2D, 0x03]

robertklep, 2025-06-15:
Very interesting! 

I don't know if this will be limited to specific models of inverter. Which model do you have?

robertklep, 2025-06-15:
The protocol is documented [here](https://github.com/robertklep/esphome-delta-solivia/blob/main/assets/Public%20RS485%20Protocol%201V2.pdf), which explains that the third byte in the sequence is "address of receiving device", and `0xFF` can be used to address _all_ devices.

So you can change that byte to a specific inverter address (mine are numbered `0x01` and `0x02`).

On page 14 the actual command that you're using (`0x0D`, _"external power control limit"_) is documented as well.

I think it would be useful to add this command in the component itself, as it seems quite useful and since it's documented properly, it wouldn't be too difficult to implement. I'll have a look later this week.

NickyNL, 2025-06-17:
That sounds even better within the component itself.
Now you mention it about the IDs I do remember reading about it some time ago. Doing it specifically for each inverter would be more neat in my use case.

I'm happy to help with testing or anything.

robertklep, 2025-06-30:
If you want to test:
```
external_components:
  - source: github://robertklep/esphome-delta-solivia
    components: [ delta_solivia ]
```

Then assign an id to the component:
```
delta_solivia:
  id: my_delta_solivia
  uart_id: solivia_uart
  ...
```

Then use, for instance, a switch to turn on/off limiting:
```
switch:
  - platform: homeassistant
    id: set_power_limit_30
    entity_id: switch.set_power_limit_30
    on_turn_on:
      - lambda: |-
          id(my_delta_solivia).set_power_limit(30);
    on_turn_off:
      - lambda: |-
          id(my_delta_solivia).set_power_limit(100);
```

Instead of using switches with a fixed percentage, you can also create a [user-defined action](https://esphome.io/components/api.html#user-defined-actions) so you can set the percentage directly from Home Assistant:
```
api:
  actions:
    - action: limit_power
      variables:
        percentage: int
      then:
        - lambda: |-
            id(my_delta_solivia).set_power_limit(percentage);
```

NickyNL, 2025-07-06:
I`ve just tested it.

Seems to work, however I had to make some changes to the switch setup to get it working:
```
switch:
  - platform: template
    name: "UGTI2 power limit 0%"
    id: ugti2_set_power_limit_0_100
    optimistic: true
    turn_on_action:
      lambda: |-
        id(my_delta_solivia).set_power_limit(0,0x03);
    turn_off_action:
      lambda: |-
        id(my_delta_solivia).set_power_limit(100,0x03);
```

This way it does work and I am able to switch on/off a single inverter.
When testing how you had setup up the switch, I could not find the switch anywhere in home assistant once the esp was updated.
Not sure if I should have rebooted home assistant or what part I should have reloaded?
My code works and exposed the switch without rebooting or reloading anything.
Testing 30% on a rainy sunday would unfortunately not work :)

robertklep, 2025-07-07:
Thanks for testing!

Makes sense that `optimistic` is required, since it's not a real switch. Perhaps the `assumed_state` setting is useful as well here. I haven't tested the code myself yet so didn't spot those requirements.

I will close this issue as completed, I'll update the README and tag the code as a new release.

robertklep, 2025-07-07:
Oh wait, I misunderstood that you had to change the type to `template` instead of `homeassistant`. I'll have to look at that a little more to see why it doesn't work with the latter.