robertklep/esphome-delta-solivia #5: [Errno 104] Connection reset by peer - after updating sensors of inverter type 53

NickyNL, 2025-06-27:
So I started working on my second group of Soliva inverters.
The first consists of 3 type 15`s.
This second group has 1 type 15, and one type 53.

The esp seems to crash/reboot, or at least loses it`s connection after each update of the type 53 inverter.
When I remove this type 53 inverter from the configuration, and only readout the type 15, then everything does work as normal.

The incoming data in the log is correct.

Than at some point it also gets stuck in "INFO Successfully resolved ugti15"

Here is a example of the logging:

[17:52:47][D][DeltaSolivia:029]: INVERTER#2 - sensor 'internal_status_4', value = 0.000000
[17:52:47][D][DeltaSolivia:029]: INVERTER#2 - sensor 'supplied_ac_energy_today', value = 11062.000000
[17:52:47][D][DeltaSolivia:029]: INVERTER#2 - sensor 'runtime_today', value = 42780.000000
WARNING ugti15 @ 192.168.31.177: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for ugti15 @ 192.168.31.177
WARNING Disconnected from API
INFO Successfully resolved ugti15 @ 192.168.31.177 in 0.000s
INFO Successfully connected to ugti15 @ 192.168.31.177 in 0.015s
INFO Successful handshake with ugti15 @ 192.168.31.177 in 0.078s
[17:54:28][D][DeltaSolivia:052]: INVERTER2 - requesting update
[17:54:28][D][uart_debug:114]: >>> 02:05:02:02:60:01:85:B8:03
[17:54:28][D][uart_debug:114]: <<< 02:06:02:FF:60:01:45:4F:45:34:36:30:31:30:32:35:32:32:32:30:32:35:32:31:35:31:32:35:31:30:30:30:31:37:34:31:32:35:31:31:35:01:02:1A:00:00:00:00:23:34:00:00:00:00:23:34:00:00:00:00:00:00:00:00:00:00:23:34:00:00:00:00:00:00:00:00:00:00:01:00:02:98:01:22:00:16:00:00:00:00:00:00:00:00:00:00:00:00:00:1A:00:E7:13:84:02:91:FF:58:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:08:98:07:D0:00:32:00:32:00:00:00:00:01:95:8A:34:00:2D:AE:54:00:00
[17:54:28][D][DeltaSolivia:024]: INVERTER#2 - updating sensors
[17:54:28][D][DeltaSolivia:040]: INVERTER#2 - sensor 'part_number', value = EOE46010252
[17:54:28][D][DeltaSolivia:040]: INVERTER#2 - sensor 'serial_number', value = 220252151251000174
[17:54:28][D][text_sensor:069]: 'Inverter#5 Serial Number': Sending state '220252151251000174'
[17:54:28][D][DeltaSolivia:029]: INVERTER#2 - sensor 'SAP_date_code', value = 825374016.000000
[17:54:28][D][DeltaSolivia:029]: INVERTER#2 - sensor 'SAP_revision', value = 12597.000000
[17:54:28][D][DeltaSolivia:029]: INVERTER#2 - sensor 'Software_rev_sc_major', value = 1.000000
[17:54:28][D][DeltaSolivia:029]: INVERTER#2 - sensor 'Software_rev_sc_minor', value = 2.000000
[17:54:28][D][DeltaSolivia:029]: INVERTER#2 - sensor 'Software_rev_sc_bugfix', value = 26.000000
[17:54:28][D][DeltaSolivia:029]: INVERTER#2 - sensor 'solar_power_input_1', value = 664.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'solar_voltage_input_1', value = 290.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'solar_current_input_1', value = 2.200000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'solar_power_input_2', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'solar_voltage_input_2', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'solar_current_input_2', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'solar_power_input_3', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'solar_voltage_input_3', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'solar_current_input_3', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ac_current_phase_1', value = 2.600000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ac_voltage_phase_1', value = 231.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ac_frequency_phase_1', value = 49.959999
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ac_power_phase_1', value = 657.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ac_reactive_power_phase_1', value = 65368.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ac_current_phase_2', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ac_voltage_phase_2', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ac_frequency_phase_2', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ac_power_phase_2', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ac_reactive_power_phase_2', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ac_current_phase_3', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ac_voltage_phase_3', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ac_frequency_phase_3', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ac_power_phase_3', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ac_reactive_power_phase_3', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'max_solar_isolation_resistance_input_1', value = 2200000.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'min_solar_isolation_resistance_input_1', value = 2000000.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'ambient_temp', value = 50.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'heatsink_temp', value = 50.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'supplied_ac_energy_total', value = 26577460.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'runtime_total', value = 2993748.000000
[17:54:29][D][DeltaSolivia:040]: INVERTER#2 - sensor 'inverter_status', value = 
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'internal_status_1', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'internal_status_2', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'internal_status_3', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'internal_status_4', value = 0.000000
[17:54:29][D][DeltaSolivia:029]: INVERTER#2 - sensor 'supplied_ac_energy_today', value = 11101.000000
WARNING ugti15 @ 192.168.31.177: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for ugti15 @ 192.168.31.177
WARNING Disconnected from API
INFO Successfully resolved ugti15 @ 192.168.31.177 in 0.002s
WARNING Can't connect to ESPHome API for ugti15 @ 192.168.31.177: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.31.177', port=6053))]: [Errno 111] Connect call failed ('192.168.31.177', 6053) (SocketAPIError)
INFO Trying to connect to ugti15 @ 192.168.31.177 in the background
INFO Successfully resolved ugti15 @ 192.168.31.177 in 0.000s
INFO Successfully resolved ugti15 @ 192.168.31.177 in 0.000s
INFO Successfully resolved ugti15 @ 192.168.31.177 in 0.000s

robertklep, 2025-06-27:
Does it also crash if you configure just the type 53 inverter? I don't have a type 53 myself so haven't been able to properly validate the parser.

NickyNL, 2025-06-27:
With the logging in the original message I had it setup with only requesting the serial number of 53, as follows:

delta_solivia:
  uart_id: solivia_uart       # the id of the UART configured above
  flow_control_pin: GPIO2     # see README.md
  has_gateway: false          # see README.md
  update_interval: 10s        # see README.md
  inverters:
    - address: 2
      variant: 53
      throttle: 30s           # see README.md
      serial_number:
        name: 'Inverter#5 Serial Number'

Since it still gave logging with much more than only that serial number, it seems to be decrypting more than what`s set up, but just not using it all afterwards.

robertklep, 2025-06-28:
Yeah I noticed that as well, but it's okay, it should only have a value when there's an error.

I did check the parser, and I can't find any issues with it. However, looking at the data that is being logged by the UART integration, it's coming up short:
``
[17:54:28][D][uart_debug:114]: <<< 02:06:02:FF:60:01:45:4F:...
```

For variant 53 I expect about 260 bytes of data, but the debug line is only showing 150 bytes. The component should wait for more bytes to come in before passing everything to the parser, but there doesn't seem to be an additional read from the UART part, so that's strange.

robertklep, 2025-06-28:
Your log is basically showing just that:
```
[17:54:28][D][uart_debug:114]: >>> 02:05:02:02:60:01:85:B8:03
[17:54:28][D][uart_debug:114]: <<< 02:06:02:FF:60:01:45:4F:45:34:36:30:31:30:32:35:32:32:32:30:32:35:32:31:35:31:32:35:31:30:30:30:31:37:34:31:32:35:31:31:35:01:02:1A:00:00:00:00:23:34:00:00:00:00:23:34:00:00:00:00:00:00:00:00:00:00:23:34:00:00:00:00:00:00:00:00:00:00:01:00:02:98:01:22:00:16:00:00:00:00:00:00:00:00:00:00:00:00:00:1A:00:E7:13:84:02:91:FF:58:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:08:98:07:D0:00:32:00:32:00:00:00:00:01:95:8A:34:00:2D:AE:54:00:00
```

The first line is the component requesting an update, and the second is the response from the inverter (which then gets read [here](https://github.com/robertklep/esphome-delta-solivia/blob/8fe450b1521c3297b58c88d63c967503bd3b0b60/components/delta_solivia/delta-solivia-component.cpp#L162) and validated and processed [here](https://github.com/robertklep/esphome-delta-solivia/blob/8fe450b1521c3297b58c88d63c967503bd3b0b60/components/delta_solivia/delta-solivia-component.cpp#L26)).

Since the component validates the size of the data (the frame header contains the number of data bytes that should be in the frame), and there is no error being logged, perhaps the amount of UART data being logged is limited by ESPHome.

If you're using an RS-485 board that requires manual flow control, you first need to pull the flow control pin high, then write the request frame, then pull the flow control pin low again (the component does that [here](https://github.com/robertklep/esphome-delta-solivia/blob/8fe450b1521c3297b58c88d63c967503bd3b0b60/components/delta_solivia/delta-solivia-component.cpp#L129-L139)).

NickyNL, 2025-06-30:
So I had a little dig myself and found info on the rx_buffer_size of the uart, and the tx_buffer_size of the logger component too.
That did not seem to fix it though.

But! I ended up trying the one thing I all along told myself it couldn`t be.
For all the testing above I had the two inverters pyhsically connected, but I just choose to read one or the other out via software.
So, now when I disconnected the type 15 from the bus it does function.

Secondly, it seemed at least not to crash when I had the solivia gateway connected.
It even gave once in a while a single correct response from inverter type53.

Probably we learned that we can`t mix inverter types?
I may have to go for two RS-485 boards, or for two complete seperate setups with individual ESP`s?

NickyNL, 2025-06-30:
I`ve got the update interval set to 10sec at the moment
And not sure if it really matters, but I`ve got RS485 modules without flow control

robertklep, 2025-06-30:
Shouldn't matter, the module will then handle flow control itself.

NickyNL, 2025-08-19:
Great, your efforts are much appreciated!
I`ve installed it yesterday evenening, it seemed to work but did not have enough solar on both inverters.
This morning, they now both give out their readings so seems it indeed fixed this issue too!

I`ll keep an eye on them as well the following days.

One thing I do notice if I browse through the logging,
Is that I the logging always says timeout on inverter1 update request, 
while it always gives going to read on the update request of inverter2.
It`s not a issue because both do work, but though you might want to know it.

[08:42:46][D][DeltaSolivia:052]: INVERTER1 - requesting update
[08:42:46][D][DeltaSolivia:157]: RESPONSE - timeout
[08:41:56][D][DeltaSolivia:052]: INVERTER2 - requesting update
[08:41:56][D][DeltaSolivia:161]: GOING TO READ

robertklep, 2025-08-30:
I just pushed v3.4.0 which might also fix the timeout issues that you were having:
```
external_components:
  - source: github://robertklep/esphome-delta-solivia@v3.4.0
    components: [ delta_solivia ]
```

NickyNL, 2025-09-06:
Just tested, yes you are right that eliminates the timeout message.

It now seems to request first only inverter 1, handle that, and than later inverter 2 and handle that.
That seems fine to me! I think we can close this issue.

Like the following snippets:
[10:15:45][D][DeltaSolivia:052]: INVERTER1 - requesting update
[10:15:45][D][uart_debug:114]: >>> 02:05:01:02:60:01:85:FC:03
[10:15:45][D][uart_debug:114]: <<< 02:06:02:FF:60:01:45:4F:45:34:36:30:31:30:32:35:32:32:32:30:32:35:32:31:35:31:32:35:31:30:30:30:31:37:34:31:32:35:31:31:35:01:02:1A:00:00:00:00:23:34:00:00:00:00:23:34:00:00:00:00:00:00:00:00:00:00:23:34:00:00:00:00:00:00:00:00:00:00:01:00:00:E3:01:33:00:07:00:00:00:00:00:00:00:00:00:00:00:00:00:09:00:EA:13:88:00:E5:FF:F4:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:08:98:08:98:00:25:00:25:00:00:00:00:01:A1:B9:E1:00:2E:A8:EF:00:00
[10:15:45][I][DeltaSolivia:024]: INVERTER#2 - updating sensors

[10:15:55][D][DeltaSolivia:052]: INVERTER2 - requesting update
[10:15:55][D][uart_debug:114]: >>> 02:05:02:02:60:01:85:B8:03
[10:15:55][D][uart_debug:114]: <<< 02:06:01:96:60:01:45:4F:45:34:35:30:31:30:32:36:38:32:32:30:32:36:38:31:36:31:33:32:38:30:30:31:31:36:33:31:33:32:38:31:36:02:32:1B:02:32:1B:02:32:1B:02:32:1B:00:98:00:2C:08:98:00:20:00:00:00:1D:00:ED:02:58:13:86:00:1E:5C:94:13:88:00:00:5C:94:13:85:00:00:00:4B:00:AA:00:1E:00:00:00:F0:02:58:13:7E:13:8F:00:01:85:2F:00:00:56:DE:00:80:01:EF:0A:CF:07:D0:08:98:00:00:00:08:00:00:00:00:00:00:00:00:00:00:00:01:01:01:01:01:01:01:01:01:01:01:01:01:01:01:01
[10:15:55][I][DeltaSolivia:024]: INVERTER#1 - updating sensors

Thanks again for your efforts.

robertklep, 2025-09-06:
Thanks for testing!