robertklep/esphome-delta-solivia #6: Most sensors are not transferred to HA

it-koncept, 2025-08-26:
I have a strange problem, my inverter responds with data but most of it is not displayed in esphome/ha - I use a d1 mini. Do you have any idea how I can find out what my problem is?

Log:
[18:19:07][D][sensor:103]: 'PV1 current Power': Sending state 610.00000 W with 0 decimals of accuracy
[18:19:07][D][sensor:103]: 'PV1 total Energy': Sending state 35950.10156 kWh with 0 decimals of accuracy
[18:19:10][D][DeltaSolivia:052]: INVERTER1 - requesting update
[18:19:10][D][DeltaSolivia:161]: GOING TO READ
[18:19:10][D][DeltaSolivia:024]: INVERTER#1 - updating sensors
[18:19:10][D][DeltaSolivia:040]: INVERTER#1 - sensor 'part_number', value = EOE46010190
[18:19:10][D][text_sensor:069]: 'PV1 Part Number': Sending state 'EOE46010190'
[18:19:10][D][DeltaSolivia:040]: INVERTER#1 - sensor 'serial_number', value = 113190161040002735
[18:19:10][D][text_sensor:069]: 'PV1 Serial Number': Sending state '113190161040002735'
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'SAP_date_code', value = 825242688.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'SAP_revision', value = 12598.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'Software_rev_ac_major', value = 2.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'Software_rev_ac_minor', value = 50.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'Software_rev_ac_bugfix', value = 27.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'Software_rev_dc_major', value = 2.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'Software_rev_dc_minor', value = 50.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'Software_rev_dc_bugfix', value = 27.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'Software_rev_display_major', value = 2.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'Software_rev_display_minor', value = 50.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'Software_rev_display_bugfix', value = 27.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'Software_rev_sc_major', value = 2.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'Software_rev_sc_minor', value = 50.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'Software_rev_sc_bugfix', value = 27.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'solar_voltage_input_1', value = 227.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'solar_current_input_1', value = 2.900000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'solar_isolation_resistance_input_1', value = 2000000.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'temperature_ntc_dc', value = 37.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'solar_input_mov_resistance', value = 0.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'ac_current', value = 2.800000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'ac_voltage', value = 235.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'ac_power', value = 610.000000
[18:19:10][D][DeltaSolivia:029]: INVERTER#1 - sensor 'ac_frequency', value = 49.950001
[18:19:12][D][sensor:103]: 'PV1 current Power': Sending state 610.00000 W with 0 decimals of accuracy
[18:19:12][D][sensor:103]: 'PV1 total Energy': Sending state 35950.10156 kWh with 0 decimals of accuracy
[18:19:13][D][DeltaSolivia:052]: INVERTER1 - requesting update

yaml:
delta_solivia:
  id: my_delta_solivia
  uart_id: solivia_uart
  update_interval: 3s
  has_gateway: false
  flow_control_pin: 0
  inverters:
    - address: 1
      variant: 15
      throttle: 5s # see below
      part_number:
        name: 'PV1 Part Number'
      serial_number:
        name: 'PV1 Serial Number'
      ac_power:
        name: 'PV1 current Power'
      supplied_ac_energy_total:
        name: 'PV1 total Energy'
      supplied_ac_energy_today:
        name: 'PV1 today Energy'
      solar_voltage_input_1:
        name: 'PV1 DC Voltage'
      solar_current_input_1:
        name: 'PV1 DC Current'
      ac_voltage:
        name: 'PV1 AC Voltage'
      ac_current:
        name: 'PV1 AC Current'
      ac_frequency:
        name: 'PV1 AC Frequency'
      ac_grid_voltage:
        name: 'PV1 Grid Voltage'
      ac_grid_frequency:
        name: 'PV1 Grid Frequency'
      runtime_total:
        name: 'PV1 Runtime Total'
      runtime_today:
        name: 'PV1 Runtime Today'
      max_ac_power_today:
        name: 'PV1 AC Power Today'
      max_solar_power_input_1:
        name: 'PV1 Solar Input Power'
      temperature_ntc_dc:
        name: 'PV1 Temperatur'

it-koncept, 2025-08-26:
It has been running for a few hours and it can see 4 sensors just fine - now that you ask about it, I can actually see that it works fine with the first 4 sensors and a couple of times I have also seen one more and there it is the first 5 sensors - and it is probably because your esp32 is faster than my esp8266 that it works for you

robertklep, 2025-08-27:
I'm not sure if it's because of the ESP32 being faster, I've seen some strange update behaviour with my own setup as well, only after some time (say 30 minutes) all sensors were getting updated as expected.

I'll take a closer look at the update code.

robertklep, 2025-08-28:
I sort of forgot why `throttle` is even a thing. I added it for situations where you have a gateway, because the gateway requests updates very often (once per second even, if I remember correctly) and you want to throttle the amount of updates sent to HA.

But when you don't have a gateway, `throttle` makes no sense, because the component itself is handling the update interval.

I'll try and see if I can disable `throttle` entirely in no-gateway mode. I _think_ the issues are related to how I'm implementing the throttle, so that would be a quick fix. Eventually I'll still have to fix the throttle issues for users that do have a gateway.

it-koncept, 2025-08-28:
I haven't looked at your code as I don't think I'm that good at it myself but it sounds like an easy way to find out if the throttle is the problem. I'll just tell you my setup so you know - 2 Delta SOLIVIA 3.3 EU G3 and a HW-097 and a D1 mini - I have two more inverters if I need to use them for testing. With my own code I get data from each inverter every two seconds so the hardware can keep up

robertklep, 2025-08-28:
It looks like just enabling debug logging by itself is a big cause for issues since it appears to be incredibly slow.

I've pushed a branch for you to test:
```
external_components:
  - source: github://robertklep/esphome-delta-solivia@wip-read-loop-debug
    components: [ delta_solivia ]
```

Make sure to set the log level to INFO:
```
logger:
  level: INFO
  ...
```

it-koncept, 2025-08-28:
I will do that tomorrow as my inverters are off today 
you will get a status tomorrow :-)

it-koncept, 2025-08-29:
anytime and thanks for your work :-)